kiddin9/Kwrt #1931: e8820s无法启动

momo54181, 2024-11-15:
型号:	ZTE E8820S
平台:	ramips/mt7621
版本:	分支: 24.10-SNAPSHOT
内核: 6.6.60
生成日期:	2024-11-15 02:08:49

刷最新固件上电反复重启，貌似和[#12964](https://github.com/coolsnowwolf/lede/issues/12964)的问题一样

NianCai, 2024-12-26:
> 型号: ZTE E8820S 平台: ramips/mt7621 版本: 分支: 24.10-SNAPSHOT 内核: 6.6.60 生成日期: 2024-11-15 02:08:49
> 
> 刷最新固件上电反复重启，貌似和[#12964](https://github.com/coolsnowwolf/lede/issues/12964)的问题一样

朋友 我的机型跟你一样的，你刷 一下最新的内核：6.6.66  看看设置wifi名称及密码后能不能保存？ 我设置后保存一直转圈圈就是保存不了

momo54181, 2024-12-27:
最新的也没法启动

momo54181, 2024-12-27:
11.25.2024-kwrt-ramips-mt7621-zte_e8820s-initramfs-kernel.bin可以启动，但更新固件回出现无法启动的问题

momo54181, 2024-12-27:
@NianCai

momo54181, 2024-12-27:
使用hc5962的bootloader和固件倒是可以正常使用，就是少了个网口

NianCai, 2024-12-27:
> 最新的也没法启动

最新的可以启动，只能刷 kernel 那个固件会进系统，如果刷 factory固件 breed会提示固件太大  进入系统后 在去升级固件 用那个升级固件  问题是设置WiFi名称及密码后 不能保存

momo54181, 2024-12-28:
2024-12-24自定义编译可用

momo54181, 2024-12-28:
[image]

closed 2025-01-05 by momo54181