kiddin9/Kwrt #1943: 谁尝鲜最新内核: 6.6.63版本有没有bug?

xibeilang0616, 2024-11-25:
### **问一下尝鲜者是粗糙粗糙还是精细？**
### **有没有上一个内核版本upnp完全不兼容的大问题什么的？**
 
### **如果没有的话我就跟进升级，省的来回折腾，升级降级，折腾人**

xibeilang0616, 2024-11-26:
> 岂止是有，简直没法用，啥啥都用不了，我上个版本的mwan3也没法安装，作者把上个版本的库给删了

### **测试了一下，upnp这一版恢复正常了**

sky540, 2024-11-26:
> > 岂止是有，简直没法用，啥啥都用不了，我上个版本的mwan3也没法安装，作者把上个版本的库给删了
> 
> ### **测试了一下，upnp这一版恢复正常了**

有没有碰到什么bug

AnChiehChen, 2024-11-26:
> ### **问一下尝鲜者是粗糙还是精细？**> ### **有没有上一个内核版本upnp完全不兼容的大问题什么的？**> ### **如果没有的话，我就跟进升级，省的来回折腾，升级降级，折腾人**请问你用的什么Uboot刷的？我的怎么刷不上去，

AnChiehChen, 2024-11-26:
> ### **问一下尝鲜者是粗糙还是精细？**
> ### **有没有上一个内核版本upnp完全不兼容的大问题什么的？**
> ### **如果没有的话，我就跟进升级，省的来回折腾，升级降级，折腾人**
请问你的用什么Uboot刷的？我的怎么刷不上去。

xibeilang0616, 2024-11-26:
> > > 岂止是有，简直没法用，啥啥都用不了，我上个版本的mwan3也没法安装，作者把上个版本的库给删了
> > 
> > 
> > ### **测试了一下，upnp这一版恢复正常了**
> 
> 有没有碰到什么bug

### **我用的软件比较少，一般就三、四个，偶尔会用安装单线多拨软件玩玩网速叠加到1000兆，目前没有发现bug**
[image]

xibeilang0616, 2024-11-26:
> > ### **问一下尝鲜者是粗糙还是精细？**
> > ### **有没有上一个内核版本upnp完全不兼容的大问题什么的？**
> > ### **如果没有的话，我就跟进升级，省的来回折腾，升级降级，折腾人**
> 
> 请问你的用什么Uboot刷的？我的怎么刷不上去。

### **去年恩山论坛提供的教程，时间长了忘了哪个帖子，到恩山论坛根据路由器型号搜帖子教程**

AnChiehChen, 2024-11-26:
> > > ### **问一下尝鲜者是粗糙还是精细？**> > > ### **有没有上一个内核版本upnp完全不兼容的大问题什么？**> > > ### **如果没有的话，我就跟进升级，省的来回折腾，升级降级，折腾人**> >> >> > 请问你用什么Uboot刷的？我的怎么刷不上去。>> ### **去年恩山论坛提供的教程，时间长了忘记哪个帖子，恩山论坛根据路由器型号搜帖子教程**
好的，谢谢啦。

fhb302, 2024-11-26:
> > > > 岂止是有，简直没法用，啥啥都用不了，我上个版本的mwan3也没法安装，作者把上个版本的库给删了
> > > 
> > > 
> > > ### **测试了一下，upnp这一版恢复正常了**
> > 
> > 
> > 有没有碰到什么bug
> 
> ### **我用的软件比较少，一般就三、四个，偶尔会用安装单线多拨软件玩玩网速叠加到1000兆，目前没有发现bug**
> ![Screenshot_2024-11-26-09-48-35-308_com android browser](https://private-user-images.githubusercontent.com/56585354/389760762-575825db-bc74-40a0-a70c-a7f7f59c9644.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzI1ODgyNTgsIm5iZiI6MTczMjU4Nzk1OCwicGF0aCI6Ii81NjU4NTM1NC8zODk3NjA3NjItNTc1ODI1ZGItYmM3NC00MGEwLWE3MGMtYTdmN2Y1OWM5NjQ0LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMjYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTI2VDAyMjU1OFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTVlYWM4ZmFlNjA0YWY1OWE0MzliMDIwMDgyNjkzZGJkYTllNjg5NDJiMzNhZjAxNzA3MGNjYTFmYzRlOTBmN2UmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.SbeVW-kWW6uLApYpVYOdosOt-l0OnA00Ch8-dyqaxeA)

测试了upnp、mwan3、homeproxy，还是存在问题，多拨也用不了

xibeilang0616, 2024-11-26:
> 测试了upnp、mwan3、homeproxy，还是存在问题，多拨也用不了

### **方法不得要领，一般多拨设置后要重启路由器，多拨才能同时拨号在线，不然还是单线拨号，其他拨号口是红色故障提示**

xibeilang0616, 2024-11-26:
> > > > > > > 岂止是有，简直没法用，啥啥都用不了，我上个版本的mwan3也没法安装，作者把上个版本的库给删了
> > > > > > 
> > > > > > 
> > > > > > ### **测试了一下，upnp这一版恢复正常了**
> > > > > 
> > > > > 
> > > > > 有没有碰到什么bug
> > > > 
> > > > 
> > > > ### **我用的软件比较少，一般就三、四个，偶尔会用安装单线多拨软件玩玩网速叠加到1000兆，目前没有发现bug**
> > > > ![Screenshot_2024-11-26-09-48-35-308_com android browser](https://private-user-images.githubusercontent.com/56585354/389760762-575825db-bc74-40a0-a70c-a7f7f59c9644.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzI1ODgyNTgsIm5iZiI6MTczMjU4Nzk1OCwicGF0aCI6Ii81NjU4NTM1NC8zODk3NjA3NjItNTc1ODI1ZGItYmM3NC00MGEwLWE3MGMtYTdmN2Y1OWM5NjQ0LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMjYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTI2VDAyMjU1OFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTVlYWM4ZmFlNjA0YWY1OWE0MzliMDIwMDgyNjkzZGJkYTllNjg5NDJiMzNhZjAxNzA3MGNjYTFmYzRlOTBmN2UmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.SbeVW-kWW6uLApYpVYOdosOt-l0OnA00Ch8-dyqaxeA)
> > > 
> > > 
> > > 测试了upnp、mwan3、homeproxy，还是存在问题，多拨也用不了
> > 
> > 
> > ### **方法不得要领，一般多拨设置后要重启路由器，多拨才能同时拨号在线，多个虚拟WAN口同时显示绿色，不然还是单线拨号，其他拨号口是红色故障提示**
> 
> 重新按照您的方法测试还是不行，在23.05版本可以多拨成功，但是这个新版本几乎每个服务都有点小毛病，您方便说下您路由器型号么？

### **图片里有具体的型号**

AnChiehChen, 2024-11-26:
> > > > ### **问一下尝鲜者是粗糙还是精细？**> > > ### **有没有上一个内核版本upnp完全不兼容的大问题什么？**> > > ### **如果没有的话，我就跟进升级，省的来回折腾，升级降级，折腾人**> >> >> > 请问你用什么Uboot刷的？我的怎么刷不上去。>> ### **去年恩山论坛提供的教程，时间长了忘记哪个，帖子恩山论坛根据路由器型号搜帖子教程**>> 好的，谢谢。[image].37.png...] 是这个吗？

AnChiehChen, 2024-11-26:
> **问一下尝鲜者是粗糙还是精细？**> > > ### **有没有上一个内核版本upnp完全不兼容的大问题什么？**> > > ### **如果没有的话，我就跟进升级，省的来回折腾，升级降级，折腾人**> >> >> > 请问你用什么Uboot刷的？我的怎么刷不上去。>> ### **去年恩山论坛提供的教程，时间长了忘记哪个，帖子恩山论坛根据路由器型号搜帖子教程**>> 好的，谢谢。！[上传截图 2024-11-26 12.42![截图 2024-11-26 12 43 46](https://private-user-images.githubusercontent.com/165123036/389798619-478524be-3fe0-4bfe-8124-50cc6442e5a1.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzI1OTc0MjcsIm5iZiI6MTczMjU5NzEyNywicGF0aCI6Ii8xNjUxMjMwMzYvMzg5Nzk4NjE5LTQ3ODUyNGJlLTNmZTAtNGJmZS04MTI0LTUwY2M2NDQyZTVhMS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTI2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTEyNlQwNDU4NDdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT00YjIyODU1YzliMWE0MTQ2ZWEwZDk0ZDlmNzA1YjJlMmYyOWQ0ODViZWE4ZjNmYzMyOGRlNTg5NDFkNzFjNDI4JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.5VyOhzn7ghg6nS44jOIoNqpCWNArfNBwgHtpTgD8810).37.png...]是这个吗？

[image]

AnChiehChen, 2024-11-26:
> > > ### **问一下尝鲜者是粗糙还是精细？**
> > > ### **有没有上一个内核版本upnp完全不兼容的大问题什么的？**
> > > ### **如果没有的话，我就跟进升级，省的来回折腾，升级降级，折腾人**
> > 
> > 
> > 请问你用什么Uboot刷的？我的怎么刷不上去。
> 
> ### **去年恩山论坛提供的教程，时间长了忘记哪个帖子，到恩山论坛根据路由器型号搜帖子教程**

![Uploading 截圖 2024-11
[image]
-26 13.03.58.png…]()

AnChiehChen, 2024-11-26:
> > > > ### **问一下尝鲜者是粗糙还是精细？**> > > ### **有没有上一个内核版本upnp完全不兼容的大问题什么？**> > > ### **如果没有的话，我就跟进升级，省的来回折腾，升级降级，折腾人**> >> >> > 请问你用什么Uboot刷的？我的怎么刷不上去。>> ### **去年恩山论坛提供的教程，时间长了忘记哪个帖子，恩山论坛根据路由器型号搜帖子教程**
> 
> 好的，谢谢。

[image]

BI7PRK, 2024-11-28:
目前6.6.63版本，最大的问题是DNS（Dnsmasq）不稳定。 如果用类似 MosDNS 做前置/后置服务时会频繁出现莫明其妙的网页打不开

xibeilang0616, 2024-11-28:
> 目前6.6.63版本，最大的问题是DNS（Dnsmasq）不稳定。 如果用类似 MosDNS 做前置/后置服务时会频繁出现莫明其妙的网页打不开

是有少量的网站打不开，或者缓慢，不过以前的版本一直就有，一直存在，一般情况下MosDNS都是关闭状态