kiddin9/Kwrt #1949: RedmiAX3000无线配置没了

code-reboot-F, 2024-11-26:
[image]

xibeilang0616, 2024-11-27:
见识过这种bug，点设置里面的重启一下试一试，一般都出现

code-reboot-F, 2024-11-27:
> ### **见识过这种bug，点设置里面的重启一下试一试，一般都出现**

已经重启了，没有效果，估计是驱动问题
我查了日志，发现wlan启动出错
[image]

GuaiYiHu, 2024-11-27:
+1 蹲一个可用的固件

GuaiYiHu, 2024-11-27:
借楼问一下楼主 AX3000 用的什么版本的科学上网呢，openclash 老是 OOM 内存溢出导致网络断连

code-reboot-F, 2024-11-27:
> 借楼问一下楼主 AX3000 用的什么版本的科学上网呢，openclash 老是 OOM 内存溢出导致网络断连

ssrplus

code-reboot-F, 2024-11-27:
> +1 蹲一个可用的固件
你那边是不是也是kernel报错

code-reboot-F, 2024-11-27:
我找到了之前小米ax3600的相似问题[https://github.com/openwrt/openwrt/issues/12815](url)，推测都是因为新板系统搭配旧版驱动导致的，这是不是需要在openwrt那边反馈一下？

GuaiYiHu, 2024-11-27:
> > +1 蹲一个可用的固件
> > 你那边是不是也是kernel报错

报错一样的

code-reboot-F, 2024-11-27:
> > > +1 蹲一个可用的固件
> > > 你那边是不是也是kernel报错
> 
> 报错一样的

那我就放心了（x

AnChiehChen, 2024-11-27:
> > > > +1蹲一个可用的固件你那边是不是也是kernel报错> >> >> > 报错一样的>>那我就放心了（请问用哪家Uboot可以刷这个固件？

code-reboot-F, 2024-11-27:
> > > > > +1蹲一个可用的固件你那边是不是也是kernel报错> >> >> > 报错一样的>>那我就放心了（请问用哪家Uboot可以刷这个固件？

我当时直接拆机刷了一个，就记得是在恩山无线论下载的，，比较久远现在不知道还有没有

AnChiehChen, 2024-11-28:
> > > > > +1蹲一个可用的固件你那边是不是也是内核报错> >> >> 报错的>>那我就放心了（请问用哪家Uboot可以刷这个固件？>>我当时直接拆了机刷了，就记得在恩山无线论下载的，，比较久远，现在不知道还有没有
有具体的位址吗

code-reboot-F, 2024-11-28:
> > > > > > +1蹲一个可用的固件你那边是不是也是内核报错> >> >> 报错的>>那我就放心了（请问用哪家Uboot可以刷这个固件？>>我当时直接拆了机刷了，就记得在恩山无线论下载的，，比较久远，现在不知道还有没有
> > > > > > 有具体的位址吗
这有一个合集，不过和我当初操作有点不一样，不保证百分百可以，你在论谈里多看看吧，好像有不用拆机的方法
[[红米AX3000/CR8809/CR8808/CR8806]](https://www.right.com.cn/forum/forum.php?mod=forumdisplay&fid=171&filter=typeid&typeid=100) 一站式刷机终结贴【红米AX3000/CR8809/CR8808/CR8806/CR880X】23.11.30

code-reboot-F, 2024-11-28:
> > > > > > +1蹲一个可用的固件你那边是不是也是内核报错> >> >> 报错的>>那我就放心了（请问用哪家Uboot可以刷这个固件？>>我当时直接拆了机刷了，就记得在恩山无线论下载的，，比较久远，现在不知道还有没有
> > > > > > 有具体的位址吗

你在恩山无线论坛看看，我之前看的那个帖子找不见了，论坛里倒是有个专门的小米ax3000分区，你可以在那边找一找（一般置顶的都是刷机贴）

AnChiehChen, 2024-11-29:
> > > > > > > +1蹲一个可用的固件你那边是不是也是内核报错> >> >> 报错的>>那我就放心了（请问用哪家Uboot可以刷这个固件？>>我当时直接拆了机刷了，就记得在恩山无线论下载的，，比较久远，现在不知道还有没有
> > > > > > > 有具体的位址吗
> 
> 你在恩山无线论坛看看，我之前看的那个帖子找不见了，论坛里倒是有个专门的小米ax3000分区，你可以在那边找一找（一般置顶的都是刷机贴）
谢谢

GuaiYiHu, 2024-11-29:
参考这篇的 uboot https://zhuanlan.zhihu.com/p/7325361034
不过我已经刷回官方系统了，官方系统 + 解锁 ssh + shellcrash

AnChiehChen, 2024-12-01:
>参考这篇uboot https://zhuanlan.zhihu.com/p/7325361034> 不过我已经刷回官方系统了，官方系统+解锁ssh+shellcrash
感谢

GuaiYiHu, 2024-12-03:
官方 uboot backup https://mao.fan/article/93

utio9527, 2024-12-24:
我的cr880x用这个版本才识别出无线设备 https://dl.openwrt.ai/firmware/ipq50xx-arm/redmi_ax3000/
[image]