kiddin9/Kwrt #1953: R2S，6.6.63无法启动

strifescl, 2024-11-28:
R2S 6.6.63无法启动，还望大佬修复

Sia7ash, 2024-11-29:
> R2S 6.6.63 cannot start, hope you can fix it

same here

Thessalhydra, 2024-12-03:
12月3日的依然无法启动,估计是上游的问题吧？

Sia7ash, 2024-12-03:
> The one on December 3rd still cannot be started. I guess it’s an upstream problem?

yeah , same here , I try it everyday with new compile but not working ! I do not know how many days i have to try to make it work again!

jpsn123, 2024-12-19:
r3s也出现了  debug发现是上游uboot问题

Sia7ash, 2024-12-20:
> R3s also had debugging and found that it was an upstream uboot problem

did you fix it?

Thessalhydra, 2024-12-21:
> > R3s also had debugging and found that it was an upstream uboot problem
> 
> did you fix it?

上游回退了uboot https://github.com/coolsnowwolf/lede/commit/062dc200788c30fc85e7e4ffe005655ea8878f8d
不过12月16日的还是无法启动，等下次更新再试试吧

jpsn123, 2024-12-23:
已提交PR, 等LEDE大神合并  https://github.com/coolsnowwolf/lede/pull/13232

wdlid, 2025-02-25:
Still no fix.

Thessalhydra, 2025-04-13:
修复了吗？