kiddin9/Kwrt #1954: 版本时间20241128分支: 24.10-SNAPSHOT 内核: 6.6.63   R2S 构建新固件无法启动

strifescl, 2024-11-28:
Building images for rockchip - FriendlyARM NanoPi R2S
Packages: firewall iptables-legacy luci-app-LingTiGameAcc luci-app-ddns luci-app-openclash luci-app-uugamebooster luci-ssl-nginx luci-theme-argon miniupnpd-iptables autocore base-files bash block-mount busybox ca-bundle coremark curl dnsmasq-full dropbear ds-lite e2fsprogs fdisk fstools htop kernel kmod-drm-rockchip kmod-gpio-button-hotplug kmod-lib-zstd kmod-tcp-bbr kmod-usb-net-rtl8152 libc libgcc libustream-mbedtls logd lsblk luci-app-advancedplus luci-app-fan luci-app-fileassistant luci-app-firewall luci-app-package-manager luci-app-syscontrol luci-app-upnp luci-app-wizard luci-base luci-compat luci-lib-fs luci-lib-ipkg mkf2fs mtd nano netifd odhcp6c odhcpd-ipv6only openssh-sftp-server opkg partx-utils ppp ppp-mod-pppoe procd resolveip swconfig uboot-envtools uci uclient-fetch urandom-seed urngd wget-ssl zram-swap automount luci-app-diskman

xenstar, 2024-12-09:
> same here , my r2s won't even start , i installed old version to make it run again

I had to run an old version as well. It seems the project is dead. It used to be the best stable firmware for R2S. I don't know where the developer is and why he started supporting so many devices, and now he can't make stable firmware for a single device.

Sia7ash, 2024-12-09:
> I had to run an old version as well. It seems the project is dead. It used to be the best stable firmware for R2S. I don't know where the developer is and why he started supporting so many devices, [...]

I don't know why the developers won't fix this problem.we have to find another solution for R2S