kiddin9/Kwrt #1956: 玩客云分区太小(只有1.1g),一键扩容无效

jiugeliji, 2024-11-29:
检测到设备/dev/mmcblk1分区数6个,请备份数据后删除分区重新操作！

master-yun-yun, 2024-12-03:
这么明显的提示还要怎么办？

master-yun-yun, 2024-12-03:
这么明显的提示还要怎么办？

jiugeliji, 2024-12-03:
> 这么明显的提示还要怎么办？

openwrt安装到内置emmc,不能删除分区,也不能扩容,(小白勿喷)

lingasly, 2025-01-26:
我也遇到这个问题，哥你找到解决方法了吗？

dj56959566, 2025-03-12:
wget -U "" -O expand-root.sh "https://openwrt.org/_export/code/docs/guide-user/advanced/expand_root?codeblock=0"
chmod +x expand-root.sh
./expand-root.sh

这是一键扩本身闲置的空间。用在N1盒子不管用。怪了。第一次用在SSH下本机路由最好有开梯子

dj56959566, 2025-03-12:
> 检测到设备/dev/mmcblk1分区数6个,请备份数据后删除分区重新操作！

wget -U "" -O expand-root.sh "https://openwrt.org/_export/code/docs/guide-user/advanced/expand_root?codeblock=0"
chmod +x expand-root.sh
./expand-root.sh

这是一键扩本身闲置的空间。用在N1盒子不管用。怪了。第一次用在SSH下本机路由最好有开梯子  再重启路由

dj56959566, 2025-03-12:
> 这么明显的提示还要怎么办？
wget -U "" -O expand-root.sh "https://openwrt.org/_export/code/docs/guide-user/advanced/expand_root?codeblock=0"
chmod +x expand-root.sh
./expand-root.sh

这是一键扩本身闲置的空间。用在N1盒子不管用。怪了。第一次用在SSH下本机路由最好有开梯子  再重启路由