kiddin9/Kwrt #1959: 这几天Mihomo出故障了

xibeilang0616, 2024-12-01:
### **以前一直好好的，最近Mihomo版本出现故障了，填写链接订阅，无法网络订阅。**
### **只能用PC上的clash填写订阅成功后，再从clash里把.yml文件上传到Mihomo里，Mihomo才能正常本地使用。**

refined-fish, 2024-12-05:
> ### **以前一直好好的，最近Mihomo版本出现故障了，填写链接订阅，无法网络订阅。**
> ### **只能用PC上的clash填写订阅成功后，再从clash里把.yml文件上传到Mihomo里，Mihomo才能正常本地使用。**

确实存在问题，我还去mihomo提[issue](https://github.com/morytyann/OpenWrt-mihomo/issues/324#issue-2715054325)，结果是搞错了，原仓库没有问题😂，目前拉取最新仍然存在此问题

ceozero, 2024-12-18:
> > ### **以前一直好好的，最近Mihomo版本出现故障了，填写链接订阅，无法网络订阅。**
> > ### **只能用PC上的clash填写订阅成功后，再从clash里把.yml文件上传到Mihomo里，Mihomo才能正常本地使用。**
> 
> 确实存在问题，我还去mihomo提[issue](https://github.com/morytyann/OpenWrt-mihomo/issues/324#issue-2715054325)，结果是搞错了，原仓库没有问题😂，目前拉取最新仍然存在此问题

我艹，我还以为是mihomo的问题。在那边查了半天。。

zhengyi0533, 2024-12-25:
我也遇到了同样的问题

cosineon, 2025-01-25:
俺也一样

vicdorke, 2025-01-25:
编译的时候不要用自带的，只把依赖默认编译进去，自带的源有问题。从github自己下载包，手动上传安装。