kiddin9/Kwrt #1961: TP-Link TL-XDR6088最新固件无法启动

AnChiehChen, 2024-12-01:
openwrt ai的新固件用各种uboot都尝试了，固件刷进去了，就是启动不了，获取不了ip，进不去后台，也没有找到wifi。

Sia7ash, 2024-12-01:
> I have tried various uboots for the new firmware of openwrt ai. The firmware has been flashed, but it just won't start, can't get the IP, can't enter the background, and can't find the wifi.

it is broken for all devices

Sia7ash, 2024-12-01:
> > > I tried various uboots for the new firmware for openwrt ai. The firmware was flashed but it won't boot, won't get an IP, won't go to the background, and won't find wifi.
> > 
> > 
> > It's broken for all devices
> 
> When will it be available? Or will it never be available?

i don't know , maybe in a few days , maybe weeks , who knows ?
i have r2s and it won't even start with new build !

AnChiehChen, 2024-12-01:
> > > > 我尝试了各种uboots来为openwrt ai的新固件。固件被刷新了，但它无法启动，无法获得IP，无法进入后台，也无法找到wifi。
> > > 
> > > 
> > > 所有设备都坏了
> > 
> > 
> > 什么时候能到？还是它永远不会可用？
> 
> 我不知道，也许几天后，也许几周后，谁知道呢？我有r2s，它甚至不会从新构建开始！

对所有设备都坏了，那就是大面积故障，既然是大面积故障，都知道了问题所在，为什么不及时处理呢？

Sia7ash, 2024-12-01:
> > > > > I tried various uboots to get new firmware for openwrt ai. The firmware was flashed but it won't boot, won't get an IP, won't go to background, and won't find wifi.
> > > > 
> > > > 
> > > > All equipment is broken
> > > 
> > > 
> > > When will it arrive? Or will it never be available?
> > 
> > 
> > I don't know, maybe in a few days, maybe in a few weeks, who knows? I have r2s and it won't even start from a new build!
> 
> If all the equipment is broken, then it is a large-scale failure. Since it is a large-scale failure and we all know the problem, why not deal with it in time?

i have no idea , maybe they are trying to solve the problem or maybe they are not really care about it . we have to find another solution for us

closed 2024-12-04 by AnChiehChen