kiddin9/Kwrt #1983: 给作者的两点建议，希望kwrt定制平台越来越好

jpsn123, 2024-12-19:
作者你好，我是KWRT 3年用户，也赞助你的为爱发电，我希望KWRT越来越好，有两点建议：
1. KWRT更新策略很激进，这是一个好事，但是一般用户来说是不希望成为OpenWrt官方的测试用户的，所以希望定制选项中能够选择稳定版本
2. 希望稳定版本的ipk仓库存在时间至少以年为单位，经常因为要安装一个无法下载的ipk包而去重新定制固件

我现在有一个设备是R3S无法使用24.10启动,  一开始以为是KWRT的原因，后来发现其他设备都是正常的，debug后才发现是24.10的使用的上游U-boot模块有问题（alloc space exhausted），但是我又无法刷23.05，因为23.05的KWRT仓库已经不存在了，所以这两天我自己在手动编译23.05，这很浪费时间。

qin9125, 2024-12-20:
一样，旧固件我删除了无法回退😔

Sia7ash, 2024-12-23:
> Hello author, I am a 3-year KWRT user and also sponsor your Power for Love. I hope KWRT will get better and better. I have two suggestions:
> 
> 1. KWRT's update strategy is very aggressive, which is a good thing, but most users do not want to become official OpenWrt testers, so they hope to be able to choose a stable version in the customization options.
> 2. I hope that the stable version of the ipk repository will exist for at least a year. I often need to re-customize the firmware because I want to install an ipk package that cannot be downloaded.
> 
> I have a device that is R3S and cannot boot with 24.10. At first I thought it was due to KWRT, but later I found that other devices were normal. After debugging, I found that there was a problem with the upstream U-boot module used by 24.10 (alloc space exhausted). However, I cannot flash 23.05 because the KWRT repository for 23.05 no longer exists. So I have been manually compiling 23.05 myself in the past two days, which is a waste of time.

it is not just R3S , I have R2S and R4S and they also not booting with new compile.
I'm just waiting to anybody that reply this massage and tells me that the boot is fixed .

jpsn123, 2024-12-24:
@Sia7ash  I also have an R4S device. I just tested it with the[ latest firmware](https://dl.openwrt.ai/releases/24.10/targets/rockchip/armv8/kwrt-12.22.2024-rockchip-armv8-friendlyarm_nanopi-r4s-squashfs-sysupgrade.img.gz), and there is no problem with startup. Please check if there is any problem with your SD card.

1447262390, 2024-12-25:
赞同，如果有可以选择稳定版本的话，那真是太好了

Sia7ash, 2024-12-30:
> @Sia7ash I also have an R4S device. I just tested it with the[ latest [...]

just test the new release and it boots again but after a moment it restarts
the sd card is ok , it runs previous versions just ok .

AnChiehChen, 2025-01-01:
tplink xdr 6088也是能刷入，启动不了。原因不详；

youhunluanma, 2025-01-11:
题主好，我最近新买的R3S使用本站系统也是无法开机，issue刚才已经详细提了（#2019），但是我在immortalwrt网站下载了目前的24.10.0-rc3版本，下载后可以正常刷入启动开机，下载本站的定制系统或者纯净系统，就是无法启动，我个人依然怀疑是现在的kwrt有问题，毕竟我同一台机器认同样的方法刷imwrt就能开机使用了。我的操作步骤详细已经写在issue#2019了，题主要是有时间的话，可以帮忙试试现在编译的能不能启动，如果题主那边能启动，希望指点指点我的问题出在哪了，我也很喜欢这家的系统，imwrt用不习惯，谢谢！
（我的R3S是2+32G）

jpsn123, 2025-01-13:
@youhunluanma   哎呦  ImmortalWrt 还真开始支持R3S了，刚刚我还不信来着。
KWRT上游是LEAN 无法启动的问题不是KWRT作者的锅，不过我是可以正常启动的，但是R3S最近版本有一个问题，就是LAN和WAN口反了，你看看换一个接口试试，这个也是上游的锅。

youhunluanma, 2025-01-13:
> @youhunluanma 哎呦 ImmortalWrt 还真开始支持R3S了，刚刚我还不信来着。 KWRT上游是LEAN 无法启动的问题不是KWRT作者的锅，不过我是可以正常启动的，但是R3S最近版本有一个问题，就是LAN和WAN口反了，你看看换一个接口试试，这个也是上游的锅。

抱歉我刚才又试了一下，交换WAN/LAN后可以正常开机进后台了。我刚才认为不能进后台原因是他开机太慢了（要开三五分钟，但我只加了不到10个插件且没有Docker，我认为这个速度是不正常的），而且进后台以后正常上网以后，他的LAN/WAN网口依然只亮一个灯、WAN/LAN指示灯（电源灯旁边的）也依然不亮，让我误以为是没有进系统。谢谢！

jpsn123, 2025-01-15:
@youhunluanma  LEAN那边的uboot设备树定义有问题，导致灯对应的不正确，不仅仅是灯，Reset按钮也用不了，但是瑞芯微官方的就是正常的，不过这些都不影响使用，有时间提交一个PR修复一下，最近工作上太忙。爱折腾你也可以学习一下他们的代码，看代码是解决问题的最快方法。