kiddin9/Kwrt #1986: 请问：分支: 24.10.0-rc3 内核: 6.6.66 有没有问题？

xibeilang0616, 2024-12-22:
### **如果有第一批尝鲜的朋友，发现bug或重大翻车事故，请及时通知一下大家，有效防止连环装车事故，哈哈😃**

AnChiehChen, 2024-12-23:
更新的这么快，反应的问题又不能及时解决，不知道这个网站的作者打算如何长期运营下去；

jaysonyip, 2024-12-23:
有，不知道为啥HomeProxy用不了，显示
TypeError
parts[i].startsWith is not a function

windows0041, 2024-12-23:
觉得更新慢的话,可以甩 张元子过去,激励一下作者

windows0041, 2024-12-23:
觉得更新慢的话,可以甩 张元子过去,激励一下作者. 有钱才有动力

NianCai, 2024-12-24:
E8820S 路由器的最新固件有问题，设置完路由器名称及密码后 不能保存，点 保存 一直在那转圈圈
[image]

NianCai, 2024-12-24:
E8820S 路由器的最新固件有问题，设置完路由器WiFi名称及密码后 不能保存，点 保存 一直在那转圈圈
[image]

NianCai, 2024-12-25:
E8820S 路由器的最新固件有问题，设置完路由器WiFi名称及密码后 不能保存，点 保存 一直在那转圈圈 求大佬尽快解决
[image]

Yuzhii0718, 2024-12-25:
@NianCai 可以删除当前的配置，然后点击添加，绑定到 LAN，这样2.4G是正常的，而5G有时候会找不到信号（与信道无关），提示未关联，应当是还没适配好了。

NianCai, 2024-12-26:
> @NianCai 可以删除当前的配置，然后点击添加，绑定到 LAN，这样2.4G是正常的，而5G有时候会找不到信号（与信道无关），提示未关联，应当是还没适配好了。

你没明白我的意思 ，我的意思的设置wifi名称及密码 不能保存，有WiF信号 2.4G 5G都有也能接收到
[image]

wjcroom, 2024-12-26:
目标平台 | mediatek/filogic
固件版本 | Kwrt 24.10.0-rc4 12.24.2024 / LuCI openwrt-24.10 branch 25.359.56502~2da1efb
 
今晚更新的rc4,
wifi密码不能保存的问题已经解决
,因为cbi.js存在隐忧.

NianCai, 2024-12-27:
> 目标平台 | mediatek/filogic 固件版本 | Kwrt 24.10.0-rc4 12.24.2024 / LuCI openwrt-24.10 branch 25.359.56502~2da1efb
> 
> 今晚更新的rc4, wifi密码不能保存的问题已经解决 ,因为cbi.js存在隐忧. @Yuzhii0718 删除确实好用,现在已经好了. 更新了js页面的一个异常.但是纳闷

为什么我更新到rc4 wifi密码还是不能保存呢
[image]

NianCai, 2024-12-27:
> @NianCai 可以删除当前的配置，然后点击添加，绑定到 LAN，这样2.4G是正常的，而5G有时候会找不到信号（与信道无关），提示未关联，应当是还没适配好了。
手动添加我也试了的，也是不行

momo54181, 2024-12-27:
https://github.com/kiddin9/Kwrt/issues/1931 E8820S无法启动

Yuzhii0718, 2024-12-28:
> #1931 E8820S无法启动

需要在 breed 下安装 initramfs，或者全量固件，在线升级则无法启动

wjcroom, 2024-12-28:
> > @NianCai 可以删除当前的配置，然后点击添加，绑定到 LAN，这样2.4G是正常的，而5G有时候会找不到信号（与信道无关），提示未关联，应当是还没适配好了。
> > 手动添加我也试了的，也是不行
可能需要，换个浏览器。缓存的js文件闹的。 
需要手工改一下， /etc/config/wireless
，高级设置[
](https://private-user-images.githubusercontent.com/111072/398610385-bd7909b9-e3fd-4ce0-b48b-f8672b31e972.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzUzODYxMzcsIm5iZiI6MTczNTM4NTgzNywicGF0aCI6Ii8xMTEwNzIvMzk4NjEwMzg1LWJkNzkwOWI5LWUzZmQtNGNlMC1iNDhiLWY4NjcyYjMxZTk3Mi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjI4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIyOFQxMTM3MTdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lOGI1MDA2MDdiMzc5YmQxMTFlOTMwMzQ1MzE5Y2M1MjJlZDI2MzkyZjk0ZDE0MWY4MTVmYmYyNGU4NTgyYjMyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.Cv6u8OzojpXWUt2AL2aRIqmq81KPjq4I4z9GlHqDLcc)

Thessalhydra, 2024-12-29:
r2s还是启动不了啊

NianCai, 2024-12-29:
多谢大佬 在最最最新版本中解决了WIFI名称及密码修改保存