kiddin9/Kwrt #1989: 无线设置无法保持

wjcroom, 2024-12-24:
无线接口设定会卡死,反复转圈无响应,继续用回原来的参数,好像是某些参数无法通过,1年前也是一样的问题.后转去了imm. 
设备CMCC RAX3000M (NAND version)
[image]

后用ssh, 手工改 /etc/config/wireless, 相应的配置才解决加密码的问题.但是这个问题已经困扰很久了.

NianCai, 2024-12-25:
我的 E8820S 路由器更新到最新版本 也是这样的，设置WiFi名称及密码 无法保存 一直转圈圈
[image]

wjcroom, 2024-12-25:
[98262.686432] mtdblock: MTD device 'bl2' is NAND, please consider using UBI block devices instead.
[98262.710403] mtdblock: MTD device 'fip' is NAND, please consider using UBI block devices instead.
[98262.758292] mtdblock: MTD device 'u-boot-env' is NAND, please consider using UBI block devices instead.
[98322.037143] mtdblock: MTD device 'ubi' is NAND, please consider using UBI block devices instead.
[98322.080243] mtdblock: MTD device 'factory' is NAND, please consider using UBI block devices instead.
[98322.123560] mtdblock: MTD device 'bl2' is NAND, please consider using UBI block devices instead.
[98322.147498] mtdblock: MTD device 'fip' is NAND, please consider using UBI block devices instead.
[98322.196388] mtdblock: MTD device 'u-boot-env' is NAND, please consider using UBI block devices instead.

xibeilang0616, 2024-12-25:
### **同样的机型，我这从没出现过这样的情况，检查检查你定制的固件有没有问题？**

### **退出登录，清理浏览器缓存，用该网站底下无定制的懒人包固件刷一刷，试一试就知道了**

wjcroom, 2024-12-26:
[image]
这里也是临时解决办法, 修改保存,重启.
@xibeilang0616  ,请问您改过分区文件mtdblock: 里面的内容吗,我安装的是定制的boot..怀疑和它有关系.
如图这样修改ap, 也无法不重启更新成新的配置.
问题在于,1,不去更新 /etc/config/wireless文件. 
2.UBI存在一些问题,无法调用后台接口,

wjcroom, 2024-12-26:
Kwrt 24.10.0-rc3 12.22.2024 / LuCI openwrt-24.10 branch 25.354.00587~f92e072
--
这是个由来已久的问题,有一年多了.  刚才测试,添加新热点,删除新热点,只要用默认的配置,都可以保存应用,生效.
问题就出在,热点编辑的页面,保存按钮,未正确执行任务.

我先分析一下源码,看可以找到地方吗. 日志也没记录有用的信息.

wjcroom, 2024-12-26:
@NianCai  经过测试, 不要使用编辑功能,使用添加菜单,可以保存. 编辑有问题,
添加成功,保存应用以后,删除默认的热点信息,应该可以回避这个问题.
@xibeilang0616  确实无法edit原有的结点,默认构建,始终这样

wjcroom, 2024-12-26:
[image]

wjcroom, 2024-12-26:
浏览报错  ,uci.js  38行
uci.js?v=25.321.80622~a403707:38 Uncaught (in promise) TypeError: Cannot read properties of undefined (reading 'disabled')
    at ClassConstructor.get (uci.js?v=25.321.80622~a403707:38:30)  

`if(d[conf][sid]===true||d[conf][sid][opt])
return null;if(c[conf]?.[sid][opt]!=null)`

这里存在一个逻辑错误, ||后面的  opt,无法引用.我修改了试试.

xibeilang0616, 2024-12-26:
> ![无标题](https://private-user-images.githubusercontent.com/111072/398610385-bd7909b9-e3fd-4ce0-b48b-f8672b31e972.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzUxODAyNDgsIm5iZiI6MTczNTE3OTk0OCwicGF0aCI6Ii8xMTEwNzIvMzk4NjEwMzg1LWJkNzkwOWI5LWUzZmQtNGNlMC1iNDhiLWY4NjcyYjMxZTk3Mi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjI2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIyNlQwMjI1NDhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lYmMyMTE4ODgxYzIyODYwNTJlNjNhOTY1ZDdhOTA5OWMxMjg0M2Q3ZWI4YjY4MmIzYWU0YzgwYzMwZDI1ODhkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.-PQgMJ8HPN7dCRnzWFgqHpGyMtDmw1dV6HKnLYSwxTM) 这里也是临时解决办法, 修改保存,重启. @xibeilang0616 ,请问您改过分区文件mtdblock: 里面的内容吗,我安装的是定制的boot..怀疑和它有关系. 如图这样修改ap, 也无法不重启更新成新的配置. 问题在于,1,不去更新 /etc/config/wireless文件. 2.UBI存在一些问题,无法调用后台接口,

我从来不去修改那些自己都不太明白的文件，

wjcroom, 2024-12-26:
@xibeilang0616   这个属于个人爱好
@NianCai  原因在uci.js一处调用异常没有加 空值检测. 这是rust一直怒v里避免的.
```
vi /www/luci-static/resources/uci.js
 return null;if(c[conf]?.[sid][opt]!=null)
修改为
 return null;if(c[conf]?.[sid]?.[opt]!=null)
```
我认为也许可以临时关闭这个issue.了. 作者怎么看这个问题

wjcroom, 2024-12-26:
闲来无事更新成了这个版本
主机名 | Kwrt
-- | --
型号 | CMCC RAX3000M (NAND version)undefined
目标平台 | mediatek/filogic
固件版本 | Kwrt 24.10.0-rc4 12.24.2024 / LuCI openwrt-24.10 branch 25.359.56502~2da1efb

 虽然问题已经不存在.但是cbi.js已经完全改了模样, 变成了未编译的. 可以看着很舒服的.
i没有以前的逻辑语句,精简版了.

closed 2024-12-26 by wjcroom

NianCai, 2024-12-27:
> @xibeilang0616 这个属于个人爱好 @NianCai 原因在uci.js一处调用异常没有加 空值检测. 这是rust一直怒v里避免的.
> 
> ```
> vi /www/luci-static/resources/uci.js
>  return null;if(c[conf]?.[sid][opt]!=null)
> 修改为
>  return null;if(c[conf]?.[sid]?.[opt]!=null)
> ```
> 
> 我认为也许可以临时关闭这个issue.了. 作者怎么看这个问题

厉害 大佬 还懂编程 会修改出错的代码

NianCai, 2024-12-27:
> ![截图20241226101055](https://private-user-images.githubusercontent.com/111072/398615914-69cd8f61-cc1d-4d1c-9f1a-cda34ae5769b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzUzMDQ1NTksIm5iZiI6MTczNTMwNDI1OSwicGF0aCI6Ii8xMTEwNzIvMzk4NjE1OTE0LTY5Y2Q4ZjYxLWNjMWQtNGQxYy05ZjFhLWNkYTM0YWU1NzY5Yi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjI3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIyN1QxMjU3MzlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT04ZTI2YzJhYjRhZGMxYmFkOWRiZmE0MWE2OGI4ZGRlMTA0NjlhZWM3MWFhNWEyZjM2ZjRlMmYwNjQ0OGZmODY5JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.N8smSPxPoaypUQgagt_wVfMIl_rnLYaQqx15fgh-u6Y)

添加的 这个无线 好像无法开启 不知道是不是我操作不对

NianCai, 2025-01-04:
> 闲来无事更新成了这个版本
> 
> 主机名	克尔特
> 型号	CMCC RAX3000M （NAND version）未定义
> 目标平台	联发科/Filogic
> 固件版本	Kwrt 24.10.0-rc4 12.24.2024 / LuCI openwrt-24.10 分支 25.359.56502~2da1efb
> 虽然问题已经不存在.但是cbi.js已经完全改了模样， 变成了未编译的.可以看着很舒服的. i没有以前的逻辑语句，精简版了.

刚看一下大佬的github主页 看到大佬的有ruby写的项目，大佬还会ruby