kiddin9/Kwrt #2001: openwrt.ai网站突发故障，无法访问

digiw, 2025-01-02:
[image]

liaol99, 2025-01-02:
hosts 152.53.108.118 openwrt.ai

hsy5571615, 2025-01-02:
> hosts 152.53.108.118 openwrt.ai

也不行

连接到 152.53.108.118 时发生错误。SSL 对等端没有所请求 DNS 名称的证书。

错误代码：SSL_ERROR_UNRECOGNIZED_NAME_ALERT

liaol99, 2025-01-02:
> > hosts 152.53.108.118 openwrt.ai
> 
> 也不行
> 
> 连接到 152.53.108.118 时发生错误。SSL 对等端没有所请求 DNS 名称的证书。
> 
> 错误代码：SSL_ERROR_UNRECOGNIZED_NAME_ALERT

那就不知道了，我能打开dl.openwrt.ai ip是152.53.108.118 然后改hosts之后就正常了

hsy5571615, 2025-01-02:
> > > hosts 152.53.108.118 openwrt.ai
> > 
> > 
> > 也不行
> > 连接到 152.53.108.118 时发生错误。SSL 对等端没有所请求 DNS 名称的证书。
> > 错误代码：SSL_ERROR_UNRECOGNIZED_NAME_ALERT
> 
> 那就不知道了，我能打开dl.openwrt.ai ip是152.53.108.118 然后改hosts之后就正常了

你用的是啥浏览器啊？我用chrome，firefox和手机浏览器都不行

systemadmin2024, 2025-01-02:
这几天断断续续打不开

AnChiehChen, 2025-01-03:
在开梯子的情况下，将openwrt.ai域名加入直连列表就可以打开了。

systemadmin2024, 2025-01-03:
> 在开梯子的情况下，将openwrt.ai域名加入直连列表就可以打开了。

直连确实可以，感谢。

AnChiehChen, 2025-01-03:
> > 在打开梯子的情况下，将openwrt.ai域名加入直连列表就可以打开了。>> 直连确实可以，谢谢。
不客气

yydjh, 2025-01-03:
要散伙了吗？

digiw, 2025-01-03:
> 在开梯子的情况下，将openwrt.ai域名加入直连列表就可以打开了。

确实可以，并且似乎不能用Cloudflare解析dns

closed 2025-01-05 by digiw