kiddin9/Kwrt #2002: 最新版本BUG反馈：zerotier插件错误

litanwei, 2025-01-03:
最新版本全新刷机安装后，zerotier插件显示未启用，ssh链接修改/etc/config/zerotier文件，把option enabled '0'的0改为1后虽然显示为：ZeroTier 运行中，但是无法界面配置和修改。
应该是更新版本的zerotier错误导致的，建议尽快查阅修复。

zhengyi0533, 2025-01-03:
卸载了界面再重新安装

windows0041, 2025-01-08:
cat > /etc/config/zerotier <<EOF
config zerotier 'sample_config'
        option enabled '1'
        list join 'ffffffffffffff'
        option nat '1'
EOF

有时间在这发言,可以多去看看zerotier的项目代码

chmdir, 2025-01-11:
> cat > /etc/config/zerotier <<EOF config zerotier 'sample_config' option enabled '1' list join 'ffffffffffffff' option nat '1' EOF
> 
> 有时间在这发言,可以多去看看zerotier的项目代

mir3g 设备, 开启zerotier 后10秒就关掉了，官网上看是连上过了的
 改这个没有里面显示的就是 1

chmdir, 2025-01-12:
> > cat > /etc/config/zerotier <<EOF config zerotier 'sample_config' option enabled '1' list join 'ffffffffffffff' option nat '1' EOF
> > 有时间在这发言,可以多去看看zerotier的项目代
> 
> mir3g 设备, 开启zerotier 后10秒就关掉了，官网上看是连上过了的 改这个没有里面显示的就是 1

我是从23.05 升级 保留配制，到24.10，zerotier有问题。   直接恢复出厂 问题解决