kiddin9/Kwrt #2007: FriendlyARM NanoPi R3S 固件不能用

yydjh, 2025-01-06:
如题，FriendlyARM NanoPi R3S 固件不能用。天灵在线编译的可以用。请大佬修复。

zhizhizz, 2025-01-07:
如果是通电后SYS灯不亮，可以试试刷老版本然后在线升级，我就是这么解决的，困扰了我三个月

lishisheng0, 2025-01-07:
哪个老版本。在线升级都不能用

Sia7ash, 2025-01-09:
> I made one myself, which can be used as a temporary substitute https://github.com/lishisheng0/openwrt-r3s/releases/download/2025.01.03-1225/openwrt-rockchip-armv8-friendlyarm_nanopi-r3s-squashfs-sysupgrade.img.gz

can you please make for R2S ?

lishisheng0, 2025-01-10:
> > I made one myself, which can be used as a temporary substitute https://github.com/lishisheng0/openwrt-r3s/releases/download/2025.01.03-1225/openwrt-rockchip-armv8-friendlyarm_nanopi-r3s-squashfs-sysupgrade.img.gz
> 
> can you please make for R2S ?

https://github.com/lishisheng0/openwrt-r2s/releases/download/2025.01.10-1312/openwrt-rockchip-armv8-friendlyarm_nanopi-r2s-squashfs-sysupgrade.img.gz
我也有个R2S不在身边，使用正常跟我说下

closed 2025-01-11 by yydjh