kiddin9/Kwrt #2014: 友善 NanoPi R5S 风扇控制问题

pluto097, 2025-01-09:
使用任何插件和脚本都无法控制主板上fan口的风扇
在istore OS中就可以控制 
在k大的固件中没有pwm目录下没有任何文件

qingtian110, 2025-01-10:
[image]
[image]

我的 r5s 有完美风扇控制。可以温控（根据温度自动控制启停、转速）、或者可以手动设置常转的风扇速度。

pluto097, 2025-01-10:
> ![图像](https://private-user-images.githubusercontent.com/87856890/402058325-f914c06c-9002-44b0-b41d-2defd0c5956a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzY1MjUzMTAsIm5iZiI6MTczNjUyNTAxMCwicGF0aCI6Ii84Nzg1Njg5MC80MDIwNTgzMjUtZjkxNGMwNmMtOTAwMi00NGIwLWI0MWQtMmRlZmQwYzU5NTZhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMTAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTEwVDE2MDMzMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWEyZTdiZWJjNWFlMzBiZjVkOTBlZjEzNTg0ZmZiYzZlY2M2ODAyMjU5N2M2Y2NiZjM0OGRlNmRlYjk1NTczYzQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.AWpTC02b_0QUuoBwoh-EdC1zBTSooanRQM3mh45C0tg) ![图像](https://private-user-images.githubusercontent.com/87856890/402059078-8bfd5aea-41cb-41d8-a4c9-7451ded423a2.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzY1MjUzMTAsIm5iZiI6MTczNjUyNTAxMCwicGF0aCI6Ii84Nzg1Njg5MC80MDIwNTkwNzgtOGJmZDVhZWEtNDFjYi00MWQ4LWE0YzktNzQ1MWRlZDQyM2EyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMTAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTEwVDE2MDMzMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTgyYTk5Y2Q3NTE3NmVhYTUyOTIyMDA1NDdiZDY5M2QxNGY0YzY2MWMyNDA2NjM2MzUyYzdhMTc1NzIzYjM2OTAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.J2QUgx8apE0QHFWVxp4Yykr3oc3Y_fqYHrk6c1zpaPI)
> 
> 我的r5s有完美的风扇控制。可以温控（根据温度自动控制启停、速度）、或者可以手动设置常转的风扇。

> ![图像](https://private-user-images.githubusercontent.com/87856890/402058325-f914c06c-9002-44b0-b41d-2defd0c5956a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzY1MjU0MTgsIm5iZiI6MTczNjUyNTExOCwicGF0aCI6Ii84Nzg1Njg5MC80MDIwNTgzMjUtZjkxNGMwNmMtOTAwMi00NGIwLWI0MWQtMmRlZmQwYzU5NTZhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMTAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTEwVDE2MDUxOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWY3NTRkYmQ4YjRmYThjNDhjMDEyODBjNDRiYWRhM2Y5ZDJkZDQ4MDU1MTNjMzczNWZiMjBiNjU3NTA4YzFkODYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.22WAwsCT71Bd-kTRe5dGdpdqtKVREdTG6YOq7oEhrk0) ![图像](https://private-user-images.githubusercontent.com/87856890/402059078-8bfd5aea-41cb-41d8-a4c9-7451ded423a2.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzY1MjU0MTgsIm5iZiI6MTczNjUyNTExOCwicGF0aCI6Ii84Nzg1Njg5MC80MDIwNTkwNzgtOGJmZDVhZWEtNDFjYi00MWQ4LWE0YzktNzQ1MWRlZDQyM2EyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMTAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTEwVDE2MDUxOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTZjZDU3NmVkZWExOTgzMGJkOGQ4NTFlZGFmMmNlYzljODdhNTg3N2RiMzg5MGYyMmQ5NTk4ODY0MDI2NjQxN2EmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.zGI0tTpU-4BG3m_47sr0dLKwXEJwoMBKuOlHOemBpBQ)
> 
> 我的r5s有完美的风扇控制。可以温控（根据温度自动控制启停、速度）、或者可以手动设置常转的风扇。

您用的是什么固件？我是openwrt.ai编译的固件

pluto097, 2025-01-11:
> cooluc 固件 https://r5s.cooluc.com/

感谢

closed 2025-01-11 by pluto097