kiddin9/Kwrt #2015: 请修复旧版软件源，谢谢

yt1988, 2025-01-10:
捐赠用户，目前使用23.05。去年发现内置的软件源全部炸掉了，到现在也只恢复了一部分。

再次感谢作者的付出，辛苦作者及时修复，否则旧版基本不可用。

`Downloading https://dl.openwrt.ai/23.05/targets/x86/64/5.15.148/Packages.gz
*** Failed to download the package list from https://dl.openwrt.ai/23.05/targets/x86/64/5.15.148/Packages.gz

Downloading https://dl.openwrt.ai/23.05/packages/x86_64/base/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_base
Downloading https://dl.openwrt.ai/23.05/packages/x86_64/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_packages
Downloading https://dl.openwrt.ai/23.05/packages/x86_64/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_luci
Downloading https://dl.openwrt.ai/23.05/packages/x86_64/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_routing
Downloading https://dl.openwrt.ai/23.05/packages/x86_64/kiddin9/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_kiddin9`

aihidao, 2025-01-11:
我也遇到了同样的问题。

gengyifeng, 2025-02-24:
6.6.47的源找不到了，能补一下么？

xRetia, 2025-03-09:
同样的问题 kmod 无法下载

GOUKI9999, 2025-03-23:
同样23.05，暂时无法升级到24，目前是targets部分还是不能用，其他packages部分倒是恢复了，导致需要 kmod之类的情况都没有源了，偏偏这部分对原版本的源依赖程度最高

jbao69, 2025-03-29:
同样问题 希望解决

taozhzzl, 2025-04-04:
同样的问题 kmod 无法下载 希望解决