kiddin9/Kwrt #2019: R3S[无法启动系统，不论SD卡或是刷入EMMC启动]，但刷入ImmortalWrt可以正常启动，已排除设备问题

youhunluanma, 2025-01-11:
一句话概括：刷入本站系统不论自定义固件，还是本站提供的纯净版，不论是SD卡还是EMMC都无法开机，但同一台机器同一张SD卡刷入ImmortalWrt是可以开机的。

具体描述如下：
新买的R3S，抱歉因为没有条件外接显示器无法看到开机情况和日志，我就详细描述操作步骤如下：

1，首先尝试定制系统刷入EMMC启动
刷入步骤是：①先SD卡启动R3S官方提供的OP23.05系统；②点击"EMMC刷入助手"选择本站定制固件刷入；③拔掉SD卡重启。
重启后电脑无法获得IP地址（已尝试重启网卡、ipconfig/release/renew），电源灯常亮，LAN/WAN指示灯均不闪烁，LAN/WAN网口灯一黄一绿；

2，然后尝试定制系统刷入SD卡启动系统
刷入步骤是：①先用R3S官方工具格式化EMMC；②选择本站定制系统使用rufus写入到SD卡（已经尝试更换过2张SD卡）；③插入SD卡到R3S，上电开机。
开机后故障同1;

3，尝试是否是个人定制插件导致问题，刷入本站提供的纯净包
操作步骤是：①退出本站账号，重启浏览器，在不登录状态打开本站，型号选择R3S，拉到页面底部，下载“选我”固件；②SD卡启动R3S官方提供的OP23.05系统；③点击"EMMC刷入助手"选择①下载的固件刷入；④拔掉SD卡重启。
开机后故障同1；

4，尝试是否是本站固件问题，更换固件测试
先测试SD卡，操作步骤是：①先用R3D官方工具格式化EMMC；②打开ImmortalWrt，下载R3S的24.10.0-rc3软件包；③使用rufus写到SD卡；④插入SD卡到R3S，插电开机
开机后，电脑可以正常获取IP，指示灯正常闪烁，上网正常
再测试EMMC，用1的方法启动R3S官方系统后，步骤②选择imwrt的系统，再重启
重启后，电脑也可以正常获取IP、指示灯正常、上网正常

遂怀疑是本站系统问题，若不是希望能获得相关帮助，限于本人技术有限，没有办法 提供开机时候的屏幕输出画面等信息，只有操作步骤。支持本站，已于去年购买永久会员，希望可有答复，谢谢！

zhizhizz, 2025-01-13:
R4S也是一样的问题，去年7月的版本都不可用。据说是上游的源码问题？

youhunluanma, 2025-01-13:
根据帖子#1983的说明，已经可以进系统了。具体可以看#1983我的回复，但同时还是希望作者（上游）能修复一下这个BUG。

xieccc, 2025-04-01:
此问题还有进展不？目前我只能通过使用 23 的版本刷完后，再升级到最新版本才能正常