kiddin9/Kwrt #2047: KWRT 新固件上传备份后管理web后台无法进入

zhangboxx, 2025-02-17:
已经试过装uhttpd了，装完之后是能进了，但是界面十分古老，不是那种内置flash的了，总感觉哪里不对，因为没有上船备份的时候后台管理界面是正常的，很美观，也进得去，这是什么新版本bug吗？
到底怎么解决

zhangboxx, 2025-02-17:
见鬼的事上传备份以后。上网还照样上，ssh后台也正常可以登陆

zhangboxx, 2025-02-22:
> 进入路由ssh， opkg update opkg install luci 就好了。

进去之后还是那种很古老的界面啊

sd1114820, 2025-02-23:
/etc/init.d/uhttpd enable 
/etc/init.d/uhttpd start

429383298, 2025-03-07:
> opkg update opkg install luci luci-i18n-chinses /etc/init.d/uhttpd enable /etc/init.d/uhttpd start

谢谢，我的就是忽然断网然后进不去主页，重启网络恢复网关还是进不去，用了你的命令好了，不知道为啥会这样，大佬知道吗。

vvnocode, 2025-03-15:
> opkg update opkg install luci luci-i18n-chinses /etc/init.d/uhttpd enable /etc/init.d/uhttpd start
大佬，是不是应该是chinese？
opkg update
opkg install luci luci-i18n-chinese
/etc/init.d/uhttpd enable
/etc/init.d/uhttpd start

madman43, 2025-04-29:
> > opkg update opkg install luci luci-i18n-chinses /etc/init.d/uhttpd enable /etc/init.d/uhttpd start
> 
> 谢谢，我的就是忽然断网然后进不去主页，重启网络恢复网关还是进不去，用了你的命令好了，不知道为啥会这样，大佬知道吗。

老毛病了……