kiddin9/Kwrt #2050: 2025-02-21版本adguardhome无法启动

cranelee, 2025-02-23:
日志记录为
Sun Feb 23 10:59:10 2025 user.err root: [luci.AdGuardHome]: Username not found in uci adguardhome

HiyungXu, 2025-02-23:
更换了luci-app-adguardhome，但是这个版本的luci不能识别adguarhome的状态，实际是可以启动的

hskun, 2025-02-24:
我可以啓動，但是無法進行攔截廣告。。。

alan21j, 2025-02-25:
請恢復到以前的版本，這個版本太多問題。

momo54181, 2025-03-02:
[luci-app-adguardhome.zip](https://github.com/user-attachments/files/19041545/luci-app-adguardhome.zip)
正常可用版本

cranelee, 2025-03-06:
改用docker运行 adguard home了，还可以开两个容器，一个走国内一个走国外。不方便的是升级固件后docker镜像需要重建

YUNYE8848, 2025-03-12:
> 改用docker运行 adguard home了，还可以开两个容器，一个走国内一个走国外。不方便的是升级固件后docker镜像需要重建

怎么搞的？

cranelee, 2025-03-14:
> > 改用docker运行 adguard home了，还可以开两个容器，一个走国内一个走国外。不方便的是升级固件后docker镜像需要重建
> 
> 怎么搞的？

host方式建两个adguard home镜像，dns和web都指定为不同端口
https://sspai.com/post/63088

hskun, 2025-03-14:
> 24.10 版本你们的adg是怎么设置的，我怎么搞都不能拦截广告。
> adguardhome	0.107.56-r1
> luci-app-adguardhome	26.069.56331~9db3555

终于知道哪里出问题了，在dns设置将强制dns转发关掉就好了