kiddin9/Kwrt #2095: 83和现在的86版本都有莫名其妙的故障

yaojogd, 2025-04-14:
我是在pve的，好像83版本开始就有错误，配置错误无法应用，重启后无法登录后台，ssh正常，不知道是什么引起的，但是网络是能正常运行，就是ui不能登陆

yaojogd, 2025-04-16:
我是重启后打开网页直接拒绝连接

AquaMT, 2025-04-16:
> 我是重启后打开网页直接拒绝连接

我是这么解决的，ssh命令行`nginx -t -c '/etc/nginx/uci.conf'`后如果有“a duplicate default server for 0.0.0.0:80 in /etc/nginx/uci.conf:59”这样子的提示，后面的59就是行数，把/etc/nginx/uci.conf这个文件的这一行所在的`server{  }` 段落删掉，例如

[image]

如果`nginx  -t`提示syntax is ok/test is successful就可以了，然后`/etc/init.d/nginx restart`重启下nginx

yaojogd, 2025-04-21:
> 我是这么解决的，ssh命令行`nginx -t -c '/etc/nginx/uci.conf'`后如果有“a duplicate default server for 0.0.0.0:80 in /etc/nginx/uci.conf:59”这样子的提示，后面的59就是行数，把/etc/nginx/uci.conf这个文件的这一行所在的`server{ }` 段落删掉，例如 [...]

你会出现配置无法保存的状况吗？？更改配置的时候会出现错误，就会出现无法保存应用，刷新后会返回未保存状态

AquaMT, 2025-04-21:
> 你会出现配置无法保存的状况吗？？更改配置的时候会出现错误，就会出现无法保存应用，刷新后会返回未保存状态

没有遇到过，出现错误的话去搜索引擎把报错信息搜一下呗？